ftrias/TeensyThreads #9: I2c OLED issue

EinarArnason, 2018-09-05:
I'm using SSD1306 OLED lcd with TeensyThreads and I have this weird problem. Anytime a thread is activated, even if it's communicating with non I2c components, it distorts the lcd buffer and the lcd shows garbage. Is this some sort of hardware limitation?

ftrias, 2018-09-05:
I believe most SSD1306 drivers store the image in memory and then upload that buffer to the card during updates. So it is likely a buffer overflow somewhere. Can you post the simplest code that will have this effect?

EinarArnason, 2018-09-06:
Turns out you're spot on. I accidentally switched counts in a 2d array so memory access was off. Turns out this library is a pretty good debugging tool :).