ftrias/TeensyThreads #13: TeensyThreads and Teensy 4

mjs513, 2019-03-12:
Hi frtas
With the up coming release of the T4 any plans to update the library for use on the T4?

Thanks
Mike

mjs513, 2019-08-04:
Playing around with lib and got it working with the T4, however, can't get setMicrosTimer working properly.  Seems to run if I do setTimeSlice, have to check that again:  https://forum.pjrc.com/threads/54711-Teensy-4-0-First-Beta-Test?p=211460&viewfull=1#post211460

nsf747, 2019-09-11:
Hey,

I'm trying to run teensyThreads on a T4 as well but i cant seem to get it to compile. Has there been any update ?

ftrias, 2019-09-11:
Can you post the errors?

lovettchris, 2019-10-07:
'IRQ_PIT_CH0' was not declared in this scope

ftrias, 2019-10-07:
Line 260 in TeensyThreads.cpp is "int old_state = currentActive;" so I think you are not using the latest TeensyThreads from github. You are probably using the TeensyThreads that came with teensyduino. It is out of date.