ftrias/TeensyThreads #14: how can I make TeensyThreads work with Adafruit M4 or even Arduino Due?

dsyleixa, 2019-03-22:
hi,
TeensyThreads looks really great providing pre-emptive multithreading.
Unfortunately  I don't own a Teensy, just Adafruit Feather M4 and Arduino Due (M3), so how can I make TeensyThreads work with Adafruit M4 or even Arduino Due?
Is there perhaps a standard Arduino portation?

ProgrammingCube, 2019-06-17:
As for porting to Adafruit M4 or Due, I don't know. But I do know that there are several AVR ports of RTOS's like ChibiOS, NucleOS, and others that you can use for the Arduino. These can do "multithreading" in the same way that TeensyThreads does, and I have used it with good success before.

dsyleixa, 2019-06-21:
admittedly I don't know how TeensyThread works, but as to RTOS I already had run into some bad issues about i's preemprive scheduler, using it with my ESP32: Although the ESP32 got 2 cores, a process is stalling totally if 1 of it's reads is stalling (like hung up e.g. in loop_forever): The scheduller can't perform a round-robin id there are no yields in 1 loop. So actuylly a "real" preemptive scheduler is needed even switching the time slices if they are busy or stalling. For my Raspberry Pi I do that by std::thread or pthread (SCHED_RR), and such a preemptive thing would be needed for Arduinos too: otherways it wouldn't be better than a co-operative MT architecture.