ftrias/TeensyThreads #15: missing file IntervalTimer.h ?

dsyleixa, 2019-03-28:
a test compilation of Blink.ino targeting adafruit Feather M4  issues a compile error, appearently the file 
IntervalTimer.h
is missing:
```
D:\arduino\portable\sketchbook\libraries\TeensyThreads-master\TeensyThreads.cpp:28:27: fatal error: IntervalTimer.h: No such file or directory

 #include <IntervalTimer.h>

                           ^

compilation terminated.

Bibliothek TeensyThreads-master in Version 1.0 im Ordner: D:\arduino\portable\sketchbook\libraries\TeensyThreads-master  wird verwendet
exit status 1
```

how can that be fixed?

ftrias, 2019-03-28:
The TeensyThreads library is designed to work only with the Teensy boards from PJRC. They are tightly tied to the software. They won't work with other boards.

closed 2019-03-28 by ftrias

dsyleixa, 2019-03-28:
that is a pity!
as the Teensy company claims that it's boards are compatible with the Arduino IDE, why can this lib not be made compatible to Arduino standards?

tonton81, 2019-03-28:
Libraries are developed off the arduino platform, but not necessarily tailored for every board on the market. The authors can’t write every specific code for everyboard based on different registers, hardware timers, and core compatibilities as every board is differently accessed and has different core to support their hardware. 

In an alternate way of thinking, why doesn’t your board support libraries intended and written for the Teensy playform?

Arduino libraries who don’t access specifics of the chips are compatible cross platform, but don’t expect libraries to access specific hardware dependancies to be cross-platform.

dsyleixa, 2019-03-28:
I see your point, but  at my (limited) level of understanding your lib uses std::threads, so a gcc (g++) compiler is supposed to compile that code to either target platform.
So what is so specific  at the Teensys that std::threads can be compiled just to M4 Teensys, and not to M4 Adafruits?

FrankBoesing, 2019-03-28:
Maybe you can read the datasheets yourself and port it to your board?
M4 is not M4 - the only identical part is the ARM-core. Atmel decided to use other periphals than NXP.

Maybe, just buying a Teensy is the easyest way to get it running.

dsyleixa, 2019-03-28:
for using my Adafruit Featherwing accessories, I would need a perfectly matching Feather Adapter for the Teensy 3.5 which is not  available yet (just for 3.1-3.2)
But back to topic, IIUC:
where is IntervalTimer.h?
is that a proprietary Teensy system file located anywhere else?

FrankBoesing, 2019-03-28:
It's in the Teensy core: https://github.com/PaulStoffregen/cores/blob/master/teensy3/IntervalTimer.h