ftrias/TeensyThreads #29: Compilation error after update Arduino 1.8.16 and TensyDuino 1.55

BOBILLIER, 2021-09-16:
Hi Fernando
Today I have updated to Arduino 1.8.16 and TensyDuino 1.55, which include last version of your library, but I encountered  a compilation error below when I try to compile your blink example:
************************************************************************************************************************
Arduino : 1.8.16 (Windows 7), TD: 1.55, Carte : "Teensy 3.6, Serial, 180 MHz, Faster, US English"
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\TeensyThreads\TeensyThreads.cpp:29:25: fatal error: utils/debug.h: No such file or directory

compilation terminated.

Erreur de compilation pour la carte Teensy 3.6
***************************************************************************************************************************

ftrias, 2021-09-16:
Please remove the offending line from TeensyThreads.cpp:

#include "utils/debug.h"

I don't know how it got there. It seems to have be put in by mistake during
a pull request.

Defragster, 2021-11-12:
TeensyDuino 1.56 in Beta - fixed code for Paul could be included for next release