ftrias/TeensyThreads #33: Threads locking with Teensy Platform 4.11

whoinow, 2022-02-22:
I have a project using TeensyThreads and when running Teensy platform 4.8, everything runs fine. The last time I tired a newer platform version (4.11) threads either straight locked or slowed down significantly. I am using platformio with VS Code as a development environment. Any thoughts on this would be appreciated. For now I am just locking my platform version to 4.8.

ftrias, 2022-02-23:
What Teensy device are you using?

whoinow, 2022-02-23:
Teensy 4.0

ftrias, 2022-02-26:
I only have a Teensy 4.1. I tried a few basic examples with PlatformIO and Teensy 4.15.0 (the latest as of today) and it all works fine. Perhaps you can share your project?

whoinow, 2022-02-26:
I will give 4.15 a shot and if I still see issues, I will share the project.