ftrias/TeensyThreads #35: Multiple Arguments at Thread Launch

SuperLaserX, 2022-07-21:
Hi Team,
I'm already using TeensyThreads for a project of mine, however, I am attempting to extend the funcionality I already have, but require launching a new thread with multiple arguments. The docs say it supports std::thread-style implementation, however you can't launch it as you would a normal std::thread(func, arg1, arg2, arg3, etc). I've tried to make it work passing my args as a pointer to a struct (void *), but have had absolutely no luck (the thread just never launches). What is the proper way to do this using TeensyThreads?

ftrias, 2022-07-21:
Passing the args as a pointer to a struct is the most reliable way to do
this. It should work. Can you post some code?

SuperLaserX, 2022-07-21:
To simplify, I'm just doing something like:
`struct a {
int b;
int c;
};

void setup() {
a *x = malloc(sizeof(a));
a->b = 0;
a->c = 1;
std::thread z(dostuff, (void *)x);
z.detach();
}

void dostuff(void *args) {
a *x = (a*)args;
println(a->b);
println(a->c);
}`

I couldn't find any examples that use multiple args on a Teensy, so I got a bit lost.

ftrias, 2022-07-21:
First line of dostuff() should be a pointer as shown below. But I suspect
it might a typo since you derefence "a" in the other lines.

void dostuff(void *args) {
a **x = (a**)args;
println(a->b);
println(a->c);
}`

Do you get any compile warnings? So if you pass one argument it works, but
if you pass a pointer it does not?

SuperLaserX, 2022-07-26:
I corrected my test app (since I hadn't built it before, I wrote it instead of dumping a novel's worth of code to point out a minor issue), and built it functional here:
`#include <TeensyThreads.h>
#include <stdint.h>
#include <stdio.h>
struct a {
int b;
byte c;
};

void setup() {
  Serial.begin(9600);
}

void dostuff(void *args) {
a *x = (a*)args;
Serial.println(x->b);
Serial.println(x->c);
delete *x;
}

void loop() {
  // put your main code here, to run repeatedly:
a *x;
x = malloc(sizeof(a));
x->b = 0;
x->c = 0x31;
std::thread z(dostuff, (void *)x);
z.detach();
}`

It appears to be functioning properly in this case. The issue I was seeing in my app was something else, and is a moot point now. Thanks so much for the prompt replies, you helped me resolve the primary issue in my application.

closed 2022-07-26 by SuperLaserX