ftrias/TeensyThreads #40: zip will not install on Mac for Arduino 2.0.3

jimdgit, 2023-02-14:
Version: 2.0.3
Date: 2022-12-05T09:29:06.685Z
Installing the zip via the Lib Manager.
Gets the error:
Error: 13 INTERNAL: Library install failed: archive is not valid: multiple files found in zip file top level

ftrias, 2024-02-27:
Using Arduino 2.3.2, I also had the same error. All I did was unzip the file and then zip it again and now it works. It appears there is something different about the ZIP file format. I uploaded the new ZIP file. In any case, I suppose there is no need to maintain Arduino 1.0 compatibility, and we should switch to using the "src" directory.

Also, keep in mind that TeensyThreads is included with the TeensyDuino distribution so there is not need to install it separately.

jimdgit, 2024-02-27:
So long ago I forgot about it bu thanks for that...