ftrias/TeensyThreads #41: Teensy 4.1 - simple code won't work

mateusz-kusmierz, 2023-05-16:
I've tried running teensyThreads, so I made below code as proof of concept, but it does not work correctly.
Whats the issue?

```cpp
#include <TeensyThreads.h>

volatile int counter1 = 0;
volatile int counter2 = 0;
volatile int counter3 = 0;

void thread1_func() {
  while(1) 
  {
    Serial.printf("thread 1 => %d\n",counter1++);
    delay(5000);
  }
}

void thread2_func() {
  while(1) 
  {
    Serial.printf("thread 2 => %d\n",counter2++);
    delay(500);
  }
}

void setup() {
  Serial.begin(115200);

  threads.addThread(thread1_func);
  threads.addThread(thread2_func);
}

void loop() 
{
Serial.printf("loop => %d\n",counter3++);
delay(2000);
}
```

Serial Monitor output:
```loop => 0
thread 1 => 0
thread 2 => 0
loop => 1
loop => 2
loop => 3
loop => 4
loop => 5
loop => 6
loop => 7
loop => 8
loop => 9
loop => 10
loop => 11
loop => 12
loop => 13
loop => 14
loop => 15
loop => 16
loop => 17
loop => 18
loop => 19
loop => 20
loop => 21
loop => 22
```

titofra, 2023-05-16:
I guess the error comes from the use of` delay ()` in a thread func. I think` delay ()` act on the main thread.
Try [threads.delay ()](https://github.com/ftrias/TeensyThreads/blob/e5ed20d5ec0633a9dded3713a4b6823ca9ebc26a/TeensyThreads.h#L292) instead.

mateusz-kusmierz, 2023-05-16:
Thank You for Your response. I've tried that but unfortunately result is exactly the same.

Merlin04, 2023-08-22:
Yeah, I just ran into this exact same issue and finally figured out that for some reason `Serial.printf` requires a larger stack than 1k - this should be documented somewhere, I think that would be a pretty common function to use in a thread.

karelv, 2023-11-01:
Is the Serial code thread-safe? Maybe use a mutex to lock the resource to the thread...

karelv, 2023-11-01:
> Yeah, I just ran into this exact same issue and finally figured out that for some reason `Serial.printf` requires a larger stack than 1k - this should be documented somewhere, I think that would be a pretty common function to use in a thread.

How do you increase the stack size on teensy 4.1?